avrdudes/avr-libc #679: [bug #59536] Add support for wchar_t type

avrs-admin, 2022-01-31:
Tue 24 Nov 2020 07:48:56 PM CET

wchar_t is not really supported by avr-libc.
However even in embedded electronics there are sometimes cases were working with wide chars becomes necessary.
To make this easier the C standard specifies a wchar.h header which provides wide variants of many other function specified in the standard library like strlen and printf

This issue was migrated from https://savannah.nongnu.org/bugs/?59536

sprintersb, 2024-03-15:
There's at least one problem though.  AVR-LibC's printf-like functions recognize `%S` as an ordinary string located in flash, as opposed to the standard that uses `%S` to specify wide strings.

Krotti83, 2025-11-06:
> AVR-LibC's printf-like functions recognize `%S` as an ordinary string located in flash, as opposed to the standard that uses `%S` to specify wide strings.

According the C99, C11 and C23 specification `%ls` (for strings) or `%lc` (for a single character) should be used for `wchar_t`. The aliases `%S` for `%ls` and `%C` for `%lc` are POSIX (SUSv2, SUSv3 and SUSv4) extensions, so far I know.