avrdudes/avr-libc #749: [patch #6161] macro change: numbering of vectors

avrs-admin, 2022-01-31:
Thu 23 Aug 2007 04:50:28 PM CEST

hello,
attached is my proposed change of the sequence of interrupt name macros. it would enable to use vector names as indexes. it is expected to allow more simply changes in the future as well.

i'm sorry i had to compress it. cause of all io-headers are involved, patch is quite large.
bye

[file #13776: numbered_vectors.diff.gz](https://savannah.nongnu.org/patch/download.php?file_id=13776)

This issue was migrated from https://savannah.nongnu.org/patch/?6161

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Wed 24 Oct 2007 01:20:26 PM CEST

Basically, I like that idea, thanks!

However, I'd like to defer the actual implementation until the
time when we are able to auto-generate the ioxxx.h header
files out of the part description XML files from Atmel.  By
that time, it'll be as simple as modifying the generator
script, and re-run it.

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 13 May 2011 12:21:40 AM CEST

Well, it's been a long time now.

The header files are now auto-generated.

All header files have <name>_vect_num definitions.

All header files have SIG_<name> definitions poisoned as deprecated.

So this patch will have to be redone, but it might be a bit easier now.

Anatoly, are you still interested in working on this patch (as you are in the "assigned" field)? Or should someone else take this up?

sprintersb, 2024-03-16:
IMO it's not a good idea to change the semantics of existing macros like in
```
 /* Timer/Counter1 Overflow */
-#define TIMER1_OVF_vect			_VECTOR(8)
-#define SIG_OVERFLOW1			_VECTOR(8)
+#define TIMER1_OVF_vect			8
+#define SIG_OVERFLOW1			8
```
Added in 8b4a247897d47a8ba37572594177df2630ae2908, each vector has now a `_num` variant, e.g. in `iob8.h`:
```
/* External Interrupt Request 0 */
#define INT0_vect_num		1
#define INT0_vect			_VECTOR(1)
#define SIG_INTERRUPT0			_VECTOR(1)

/* External Interrupt Request 1 */
#define INT1_vect_num		2
#define INT1_vect			_VECTOR(2)
#define SIG_INTERRUPT1			_VECTOR(2)
...
```
keeping the old macro definitions.  Closed this one thusly.

closed 2024-03-16 by sprintersb