avrdudes/avr-libc #765: [patch #6805] Numeric interrupt vectors

avrs-admin, 2022-01-31:
Sun 12 Apr 2009 07:26:05 AM CEST

This patch allows a C++ interrupt handler name to be represented by a number, and while the name will be mangled as usual, the number survives for later linking. The patch provides for an optional numeric argument to the signal function.

An interrupt function prototype using this system for the same USART0 receiver interrupt looks like:

void IntName(void) **_attribute_** ((signal(18), __INTR_ATTRS));

[file #17905: ClassInterrupts2.zip](https://savannah.nongnu.org/patch/download.php?file_id=17905)

This issue was migrated from https://savannah.nongnu.org/patch/?6805

avrs-admin, 2022-01-31:
Anatoly Sokolov <aesok>
Sun 12 Apr 2009 04:10:51 PM CEST

The patch for the GCC is not complete.
1. No numeric argument to the 'interrupt' attribute is implemented.
2. Not possible use one interrupt handler for some interrupt vectors.

I propose to implement the interface for interrupt handler functions as:

**_attribute_** ((vectors(N1[, N2[, ...]]), [signal, interrupt, nmi, naked]))
void handler_mame (void);

The 'vectors' attribute specify interrupt vector numbers for handler function.

The 'signal', 'interrupt', 'nmi' and 'naked' attributes specify type of handler function. Need use only one. The default attribute is  'signal'.

Anatoly.

avrs-admin, 2022-01-31:
Ron Kreymborg <ronkrem>
Mon 13 Apr 2009 03:16:53 AM CEST

Good idea - a more complete implementation that mine (which was based on Marek's original). I assume it will use the number for the vector and not the "handler_name" (the C++ requirement).

Just curious - what is an example of (2)?

Ron

sprintersb, 2024-07-28:
Fixed by https://github.com/avrdudes/avr-libc/commit/5023931a1e9fe8d3d14d4c018446ac059e9e715c.  Notice that this is only the AVR-LibC part.  The feature relies on avr-gcc [PR116056](https://gcc.gnu.org/PR116056) added in v15.

closed 2024-07-28 by sprintersb