avrdudes/avr-libc #813: [patch #8826] Add UCSR0C bit definitions for ATmega1284(p) USART in SPI mode

avrs-admin, 2022-01-31:
Tue 08 Dec 2015 04:08:10 PM CET

Headers iom1284.h and iom1284p.h are lacking definitions for UCSR0C bit names used only in SPI mode: UDORD0 (bit 2, aliases UCSZ01) and UCPHA0 (bit 1, aliases UCSZ00).

This patch adds two appropriate #defines to both files mentioned above.

[file #35666: iom1284(p)_UCSR0C.patch](https://savannah.nongnu.org/patch/download.php?file_id=35666)

This issue was migrated from https://savannah.nongnu.org/patch/?8826

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Sun 11 Jun 2017 10:51:15 PM CEST

Thinking a little about it, I'm not that convinced yet about this.

All the modern ATmega-style devices (starting with the ATmega2560)
document these bits that way.  So if at all, we'd have to patch
all their header files, to supply the UDORDn and UCPHAn bits.

However, these bits are nowhere else named that way.  In particular,
none of the Atmel device description files mention these names, so
Atmel Studio doesn't know about them either.

But again: if we would really want to change something, then for
 **all** devices that are affected, not just the ATmega1284*.

sprintersb, 2024-03-16:
Closed according to comment.

closed 2024-03-16 by sprintersb