avrdudes/avr-libc #817: [patch #8966] Update RFR devices header files

avrs-admin, 2022-01-31:
Mon 04 Apr 2016 11:01:33 AM CEST

RFR devices (mega64/644/128/1284/256/2564rfr2) headers are included in the revision 2308.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2308](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2308)

It seems they are in struct-based format like xmega headers. Usually tiny and mega headers are in non struct-based format. Atmel published headers for these RFR devices are updated. This patch updates the RFR devices header as per latest headers from Atmel.

Posted the patch in the mailing list:
 [http://lists.nongnu.org/archive/html/avr-libc-dev/2016-03/msg00005.html](https://savannah.nongnu.orghttp://lists.nongnu.org/archive/html/avr-libc-dev/2016-03/msg00005.html)

[file #36830: rfr-device-headers.patch](https://savannah.nongnu.org/patch/download.php?file_id=36830)

This issue was migrated from https://savannah.nongnu.org/patch/?8966

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Sun 11 Jun 2017 10:57:44 PM CEST

As already discussed in the mailing list, applying this patch
would essentially remove a lot of existing functionality from
the header files (it removes about 20000 lines in total), just
for the sake of "we've always been doing it that way", without
any further benefit for the users.

Therefore, I wouldn't want to apply it.

If there are any actual bugs to be fixed, they'd better be
fixed without removing unrelated pieces of code.

sprintersb, 2024-02-08:
Closed as of Jörg's comment.

closed 2024-02-08 by sprintersb