avrdudes/avr-libc #821: [patch #9405] Add ISR_NOGCCISR macro to avr/interrupt.h

avrs-admin, 2022-01-31:
Mon 10 Jul 2017 03:12:15 PM CEST

This adds new ISR_NOGCISR macro to be used as ISR attribute with a similar naming like the already existing ones.  For the feature, see

[https://gcc.gnu.org/onlinedocs/gcc/AVR-Function-Attributes.html](https://savannah.nongnu.orghttps://gcc.gnu.org/onlinedocs/gcc/AVR-Function-Attributes.html)
 [https://sourceware.org/PR21683](https://savannah.nongnu.orghttps://sourceware.org/PR21683)

```
* include/avr/interrupt.h (ISR_NOGCCISR): New ISR convenience macro.
(ISR_NOICF) [!__has_attribute]: Define as empty.

```

[file #41158: alibc-nogccisr.diff](https://savannah.nongnu.org/patch/download.php?file_id=41158)

This issue was migrated from https://savannah.nongnu.org/patch/?9405

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Wed 17 Jan 2018 04:58:29 PM CET

Ping #1

sprintersb, 2024-02-01:
Resolved by https://github.com/avrdudes/avr-libc/commit/3bb12ecaaa9038d228c3072ef34882ba40ac82df.

closed 2024-02-01 by sprintersb