avrdudes/avr-libc #826: [patch #9594] Add missing TXD0 and TXD1 pins for the ATmega324PA

avrs-admin, 2022-01-31:
Tue 13 Mar 2018 11:07:12 PM CET

The ATmega324PA has 2 U(S)ARTs. In its I/O, it has `TXD_DDR`, `TXD_PORT`, `TXD_PIN` and `TXD_BIT` for accessing the TXD of the first U(S)ART. This patch rename `TXD_*` to `TXD0_*` and also add `TXD1_*`, which corresponds to the 3rd pin of the port D.

[file #43542: add_missing_txd0_txd1.patch](https://savannah.nongnu.org/patch/download.php?file_id=43542)
[file #43543: add_missing_txd0_txd1_rxd0_rxd1.patch](https://savannah.nongnu.org/patch/download.php?file_id=43543)

This issue was migrated from https://savannah.nongnu.org/patch/?9594

avrs-admin, 2022-01-31:
Gabriel-Andrew Pollo-Guilbert <gpol>
Tue 13 Mar 2018 11:13:44 PM CET

I noticed the same is for the `RXD_*`. I added a similar patch for it.

sprintersb, 2024-03-14:
Fixed by https://github.com/avrdudes/avr-libc/commit/f07cf42166051592528e1415d8ab6d2fd289339d.

closed 2024-03-14 by sprintersb