avrdudes/avr-libc #828: [patch #9779] Support for ATMega168pb

avrs-admin, 2022-01-31:
Fri 29 Mar 2019 01:26:54 PM CET

Hi

This patch adds support for ATMega168pb. I added two variants of the patch one before running bootstrap and one after. I don't know which one you need. I actively using this patch in one of my devices, so far without any problem.

Chris

[file #46660: avr-libc-2.0.0_atmega168pb_bootstrapped.patch.bz2](https://savannah.nongnu.org/patch/download.php?file_id=46660)
[file #46659: avr-libc-2.0.0_atmega168pb.patch.bz2](https://savannah.nongnu.org/patch/download.php?file_id=46659)

This issue was migrated from https://savannah.nongnu.org/patch/?9779

sprintersb, 2022-11-28:
The following change to `configure.ac` is outdated and should be removed without replacement:
```
+CHECK_AVR_DEVICE(atmega168pb)
+AM_CONDITIONAL(HAS_atmega168pb, test "x$HAS_atmega168pb" = "xyes")
+
```
Reason is the same like in issue #892.

sprintersb, 2024-03-15:
Support added in https://github.com/avrdudes/avr-libc/commit/927b0cf93a9b20c733aa4de857b78f405bb1c5ae.

closed 2024-03-15 by sprintersb