avrdudes/avr-libc #832: [patch #10171] Fix PRIi64 and friends

avrs-admin, 2022-01-31:
Thu 20 Jan 2022 01:41:31 PM CET

- PRIi64 and friends should be provided when support for long long is
present in printf and scanf, that is
__avr_libc_does_not_implement_long_long_in_printf_or_scanf is not
defined. Apparently the double negation previously let to confusion
- Add Doxygen documentation to the defines and expose the code to
Doxygen

[file #52700: 0001-Fix-PRIi64-and-friends.patch](https://savannah.nongnu.org/patch/download.php?file_id=52700)

This issue was migrated from https://savannah.nongnu.org/patch/?10171

sprintersb, 2024-03-16:
I don't actually see the purpose of this patch, as AVR-LibC's print functions don't support 64-bit values.

sprintersb, 2024-07-29:
AVR-LibC doesn't support 64-bit printf anyway.

closed 2024-07-29 by sprintersb