avrdudes/avr-libc #833: [task #3586] Change from avr architectures to lib per device.

avrs-admin, 2022-01-31:
Wed 15 Dec 2004 12:17:55 AM CET

Currently avr-libc is designed to produce a library per avr "architecture" (avr2, avr3, avr4, avr5). This architecture is arbitrary and has no real basis at Atmel.

avr-libc need to change to produce a library per device. This would allow greater flexibility, and correct features for all devices (for example correct EEPROM routines for certain devices).

This also affects binutils and gcc. They will need to be changed to move away from these architectures to per device.

This issue was migrated from https://savannah.nongnu.org/task/?3586

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 13 May 2011 01:09:34 AM CEST

This task needs to be revisited and put in the queue to be done.

sprintersb, 2023-11-25:
IMO this can be closed. 

We DO HAVE "correct EEPROM routines for [all] devices" because they are in lib<mcu>.a which IS on device granularity just like crt<mcu>.o.

For libm and libc there is no need to be on MCU level, it's just fine to be on Core level where the differences are are like: CALL vs. RCALL, SP handling, PC size, availability of MUL, MOVW, ELPM, RAMPD, etc.

Any lib function like printf or sqrt that requires action on I/O SFRs would be very questionable.

sprintersb, 2024-02-01:
No need to do ALL of the libs on level of devices.  Closed as of comment above.

closed 2024-02-01 by sprintersb