avrdudes/avr-libc #834: [task #3593] Merge libc and libm.

avrs-admin, 2022-01-31:
Wed 22 Dec 2004 11:30:24 PM CET

Merge the C library (libc) and the math library (libm).

This issue was migrated from https://savannah.nongnu.org/task/?3593

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 13 May 2011 01:10:07 AM CEST

We need to decide whether to do this task or not. Does it still have merit?

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Fri 13 May 2011 09:04:12 AM CEST

Certainly it does, and be it only for AVR Studio never learning to add "-lm" to their default Makefile ...

sprintersb, 2023-11-25:
IMO this can be closed.

Since avr-gcc v4.7.2 (fall 2012) [PR54461](https://gcc.gnu.org/PR54461) the complier driver is adding `-lm` to `LIBGCC_SPEC`, i.e. whenever the compiler is linking against `libgcc.a`, it will also link against `libm.a` provided the compiler is configured to use AVR-LibC (which is the default). In that case, the compiler is using `avrlibc.h` which reads:
```c++
#undef  LIB_SPEC
#define LIB_SPEC                                \
  "%{!mmcu=avr1:-lc} %(avrlibc_devicelib)"

// AVR-Libc implements functions from libgcc.a in libm.a, see PR54461.
// For a list of functions which are provided by libm.a and are
// omitted from libgcc.a see libgcc's t-avrlibc.

#undef  LIBGCC_SPEC
#define LIBGCC_SPEC                             \
  "%{!mmcu=avr1:-lgcc -lm}"

#undef  STARTFILE_SPEC
#define STARTFILE_SPEC                          \
  " %(avrlibc_startfile) "

#undef  LINK_GCC_C_SEQUENCE_SPEC
#define LINK_GCC_C_SEQUENCE_SPEC \
  "--start-group %G %{!nolibc:%L} --end-group"
```
This means a user (or IDE for that matter) need not add `-lm` by hand. For example, when adding `-v` to the link options, the driver will print something like:
```none
.../collect2 [...] --start-group -lgcc -lm -lc -l<mcu> --end-group
```

sprintersb, 2024-02-01:
Closed according to previous comment: avr-gcc is adding `-lm` to the link options, which is way simpler than to merge libm into libc.

closed 2024-02-01 by sprintersb