avrdudes/avr-libc #874: ioa5795.h vector numbers are wrong by a factor of 2

stefanrueger, 2022-03-05:
The interrupt vectors in include/avr/ioa5795.h are wrong by a factor of 2.

**They are**

```
/* Transponder Mode Interrupt */
#define TPINT_vect            _VECTOR(2)
#define TPINT_vect_num        2

/* External Interrupt Request 0 */  
#define INT0_vect            _VECTOR(4)
#define INT0_vect_num        4
```
...
```
/* Store Program Memory Ready  */
#define SPMREADY_vect            _VECTOR(44)
#define SPMREADY_vect_num        44
```

**They should be...**

```
/* Transponder Mode Interrupt */
#define TPINT_vect            _VECTOR(1)
#define TPINT_vect_num        1

/* External Interrupt Request 0 */
#define INT0_vect            _VECTOR(2)
#define INT0_vect_num        2
```
...
```
/* Store Program Memory Ready  */
#define SPMREADY_vect            _VECTOR(22)
#define SPMREADY_vect_num        22

```

stefanrueger, 2022-03-05:
... and `_VECTORS_SIZE` needs to be 46, not 90. Perhaps best to drop in the newer version, say, this [`include/avr/ioa5795.h`](https://raw.githubusercontent.com/eerimoq/avr-toolchain-windows/master/avr/include/avr/ioa5795.h)

sprintersb, 2024-02-07:
It looks like avr-gcc has files this device in the wrong multilib `avr5` (progmem of 16Kib to 64KiB), but it only has 8KiB and hence should be in `avr4`. 

The header from Atmel has `_VECTORS_SIZE=46` which you say is correct.

The wrong multilib set menas that the entries in the vectab are also wrong (4-byte long) where the correct size is 2 bytes. Is that correct?

stefanrueger, 2024-02-07:
Yes, 8 kB flash, 23 ISR vectors that take up 46 bytes, 2 each.

sprintersb, 2024-02-08:
Ok, avr-gcc has it in the wrong multilib then: Currently `avr5` which means that vectab entries are 4 bytes each (because `__AVR_HAVE_JMP_CALL__` according to [avr-devices.cc](https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/config/avr/avr-devices.cc;h=456a6b72c8b0da07f6a68c194d6d9f02ea53dbf6;hb=HEAD)). 

Also the device specs from Atmel atpack from 2022 has it wrong (avr5) ?

stefanrueger, 2024-02-08:
> the device specs from Atmel atpack from 2022 has it wrong

Sure, that may be why this bug came into play for avr-libc

sprintersb, 2024-02-08:
Applied the 2022 ioa5795.h as https://github.com/avrdudes/avr-libc/commit/361a2340f05bf6b7466128a719afd5c7ba6346d7.

The vectors are still wrong because ATA5795 is in the wrong multilib set (so vectab entries will occupy 4 bytes instead of the correct 2 bytes). As far as I know, there is no way to tell AVR-LibC the correct vect sizes: It uses `__AVR_HAVE_JMP_CALL__` which is defined by avr-gcc depending on the multilib set. For the latter, see avr-gcc [PR113824](https://gcc.gnu.org/PR113824).

closed 2024-02-08 by sprintersb

sprintersb, 2024-02-08:
Also notice #943, which always uses a vector size of 2 bytes for ATA5795, irrespective of avr-gcc's multilib assignment.