avrdudes/avr-libc #877: Signature bytes wrong for a number of devices

stefanrueger, 2022-03-07:
Some `SIGNATURE_2` values are wrong in io header files.

`iom164.h` should be:

```
#if defined(__AVR_ATmega164A__)
#define SIGNATURE_2 0x0F
#else
#define SIGNATURE_2 0x0A
#endif
```

`iom324.h` should be
```
#if defined(__AVR_ATmega324A__)
#define SIGNATURE_2 0x15
#else
#define SIGNATURE_2 0x08
#endif
```

see [datasheet page 298](https://ww1.microchip.com/downloads/en/DeviceDoc/ATmega164A_PA-324A_PA-644A_PA-1284_P_Data-Sheet-40002070B.pdf)

-------------------

`iom169.h` should be
```
#if defined(__AVR_ATmega169A__)
#define SIGNATURE_2 0x11
#else
#define SIGNATURE_2 0x05
#endif
```

`iom329.h` should be
```
#if defined (__AVR_ATmega329P__) || defined (__AVR_ATmega329PA__)   
#define SIGNATURE_2 0x0B
#else
#define SIGNATURE_2 0x03
#endif
```

`iom3290.h` should be
```
#if defined (__AVR_ATmega3290P__)
#define SIGNATURE_2 0x0C
#else
#define SIGNATURE_2 0x04
#endif
```

`iom6490.h` should be
```
#if defined (__AVR_ATmega6490P__)
#define SIGNATURE_2 0x0C
#else
#define SIGNATURE_2 0x04
#endif
```

see [datasheet 169/329/p/pa, page 294](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8284-8-bit-AVR-microcontroller-ATmega169A_PA_329A_PA_3290A_PA_649A_P_6490A_P_datasheet.pdf)
and [datasheet 329p page 299](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8021-AVR-ATmega329P-3290P_Datasheet.pdf)

-------------------

`iom325.h` should be
```
#if defined (__AVR_ATmega325P__)
#define SIGNATURE_2 0x0D
#else
#define SIGNATURE_2 0x05
#endif
```

`iom645.h` should be
```
#if defined (__AVR_ATmega645P__)
#define SIGNATURE_2 0x0D
#else
#define SIGNATURE_2 0x05
#endif
```

`iom3250.h` should be
```
#if defined (__AVR_ATmega3250P__)
#define SIGNATURE_2 0x0E
#else
#define SIGNATURE_2 0x06
#endif
```

`iom6450.h` should be
```
#if defined (__AVR_ATmega6450P__)
#define SIGNATURE_2 0x0E
#else
#define SIGNATURE_2 0x06
#endif
```

see [datasheet 165/325/p/pa, page 264](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8285-8-bit-AVR-Microcontroller-ATmega165A_PA_325A_PA_3250A_PA_645A_P_6450A_P_datasheet.pdf)
and [datasheet 325p page 274](https://ww1.microchip.com/downloads/en/DeviceDoc/doc8023.pdf)

dl8dtl, 2022-03-07:
> `#if defined (__AVR_ATmega325P__)`

This lacks the ATmega325PA which also needs 0x0D (see ATDF file).

stefanrueger, 2022-04-08:
More wrong signature bytes:

`iotn4.h` and `iotn5.h` should define `SIGNATURE_1` to `0x8F` (not `0x90`). See .atdf files and avrdude signature entries.

------------------------------------------------------------------

`io4434.h` should define `SIGNATURE_1` to `0x92` (not `0x93`). See avrdude signature entries (consistent with 0x92 typically implying 4 kB flash).

It also should define `SIGNATURE_2` to `0x02` (not `0x03`). See avrdude signature entries. The signatures that are currently defined for the `AT90S4434` are actually those from the `AT90S8535`.

------------------------------------------------------------------

Following the change of signature in avrdude for the `ATmega165`, the file `iom165.h` should now define `SIGNATURE_2` as `0x10`. See [avrdude issue 899](https://github.com/avrdudes/avrdude/issues/897) and [commit](https://github.com/avrdudes/avrdude/commit/75e4d064637d3e19a5010a2f71e11de5c39c278b).

sprintersb, 2025-08-11:
What remains to be done is verifying (or fixing) the signatures for:
`atxmega16c4 atxmega32c4 atxmega16e5 atxmega32a4`

For a discussion, see https://github.com/avrdudes/avrdude/issues/2048

sprintersb, 2025-08-11:
With the last patch the signature → MCU mapping looks reasonable.
```
0x1e9405: atmega169pa, atmega169p, atmega169
0x1e9407: atmega165pa, atmega165, atmega165p
```