avrdudes/avr-libc #884: atmega4809 not supported?

avrf0, 2022-05-26:
/usr/bin/avr-ld: cannot find crtatmega4809.o: No such file or directory
/usr/bin/avr-ld: cannot find -latmega4809:  No such file or directory
These files should be in my case in /usr/avr/lib/avrxmega3/
attiny x16 are working correctly.

Scrappers-glitch, 2022-06-20:
The `atmega4809` isn't listed on the official avr docs see [avr-libc](https://www.nongnu.org/avr-libc/user-manual/), but other atmega48 are there, but there are examples on [Microchip github](https://github.com/MicrochipTech/Getting_Started_with_GPIO) [Microchip support](https://github.com/microchip-pic-avr-examples/atmega4809-getting-started-with-usart-studio), idk it may be the same core as `atmega48` family, you have to do a deep search on microchip page and the datasheet, this might be a little bit tricky, though FYI defining a wrong chip in avr dude will still upload the code if the chip has the same flash size but if the internal registers layout is different and you are using something chip specific, it will raise a compile time error as it's not defined, so the avrdude flag just defines the chip specific macros, nothing more, so i anticipate that there are chip specific macros for atmega4808/4809 over atmega48.

sprintersb, 2024-01-31:
Fixed by afb1ffdb8e35065f7fb5317bb6693311f5eec7a5

closed 2024-01-31 by sprintersb