avrdudes/avr-libc #886: `crtatmega*.o:(.init9+0x0): undefined reference to main` when compiling with clang

KOLANICH, 2022-06-08:
Hi. I have modified a well-known set of CMake scripts to allow compiling with clang. Unfortunately in the end I get 

```
crtatmega2560.o:(.init9+0x0): undefined reference to `main'
```

I guess something may be flawed, either these `.o` files, or my flags.

```
readelf -s ./arduino_ide/1.8.19/hardware/tools/avr/avr/lib/avr6/crtatmega2560.o | grep main
    75: 00000000     0 NOTYPE  GLOBAL DEFAULT  UND main

readelf -s ./my_app.o | grep main
   111: 000014e8    66 FUNC    GLOBAL DEFAULT    2 main
```

What are these files, what do they do and can they be generated from `atdf`s?

sprintersb, 2022-12-03:
Usually, the problem is that you are linking without `my_app.o`. So maybe check the link command that's issued by your system.

`crtatmega2560.o` contains (parts of) the start-up code for ATmega2560, and it is supposed to be able to link against `main` (and `_exit`) for that matter.

KOLANICH, 2022-12-03:
Thanks for the description. I guess I should have closed thi issue long ago, I just have forgotten about its existance. The prkblem was with the code in CLang used as a linker frontend. I have created a fix and everytying has started be built fine, if I remember right.

KOLANICH, 2022-12-03:
https://github.com/llvm/llvm-project/commit/2d01a8572076a93fa40c4af934e78a859f3096ee

closed 2022-12-03 by KOLANICH

KOLANICH, 2022-12-03:
But I still wonder if these files SDK-shipped files can be automatically generated anyhow.