avrdudes/avr-libc #890: cannot find crtatmega4809.o

rwirth, 2022-07-15:
Trying to link any program for the mega4809 fails with

    $ avr-gcc -x c -mmcu=atmega4809 - <<<"int main() {return 0;}"
    /usr/bin/avr-ld: cannot find crtatmega4809.o: No such file or directory
    /usr/bin/avr-ld: cannot find -latmega4809: No such file or directory

It seems the device-specific libs do not get generated during build due to a typo in `gen-avr-lib-tree.sh`, lines 354ff:

```
attiny1608:crtm1608.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
attiny1609:crtm1609.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
attiny3208:crtm3208.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
attiny3209:crtm3209.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
attiny4808:crtm4808.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS};\
attiny4809:crtm4809.o:${DEV_DEFS}:${CFLAGS_SPACE}:${DEV_ASFLAGS}\
```

The device names should be atmega, not attiny.

rwirth, 2022-07-20:
Also, the ATmega808/809 lines are missing.

sprintersb, 2024-01-31:
Fixed by afb1ffdb8e35065f7fb5317bb6693311f5eec7a5

closed 2024-01-31 by sprintersb