avrdudes/avr-libc #899: build fails saying configure: error: cannot compute suffix of object files: cannot compile

bkerin, 2022-10-24:
It seems like gcc is running the wrong assembler somehow.

This failure happens fairly early in the build process, here is full context:

```
$ ./configure --prefix=/home/bkerin/opt/avr_toolchain --build=`./config.guess` --host=avr
checking build system type... x86_64-unknown-linux-gnu
checking host system type... avr-unknown-none
checking if configuring for cross compile... yes
checking if target host is avr... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for avr-strip... avr-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for avr-gcc... avr-gcc
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/bkerin/tmp/avr-libc-2.1.0':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
```

From config.log:

```
Configured with: ../configure --prefix=/home/bkerin/opt/avr-gcc-
12.2.0/ --target=avr --enable-languages=c --disable-nls --disabl
e-libssp
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.2.0 (GCC) 
configure:3547: $? = 0
configure:3536: avr-gcc -V >&5
avr-gcc: error: unrecognized command-line option '-V'
avr-gcc: fatal error: no input files
compilation terminated.
configure:3547: $? = 1
configure:3536: avr-gcc -qversion >&5
avr-gcc: error: unrecognized command-line option '-qversion'; di
d you mean '--version'?
avr-gcc: fatal error: no input files
compilation terminated.
configure:3547: $? = 1
configure:3567: avr-gcc -o conftest    conftest.c  >&5
as: unrecognized option '-mmcu=avr2'
configure:3571: $? = 1
configure:3784: checking whether we are cross compiling
configure:3822: result: yes
configure:3826: checking for suffix of object files
configure:3848: avr-gcc -c   conftest.c >&5
as: unrecognized option '-mmcu=avr2'
configure:3852: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "avr-libc"
| #define PACKAGE_TARNAME "avr-libc"
| #define PACKAGE_VERSION "2.1.0"
| #define PACKAGE_STRING "avr-libc 2.1.0"
| #define PACKAGE_BUGREPORT "avr-libc-dev@nongnu.org"
| #define PACKAGE_URL ""
| #define PACKAGE "avr-libc"
| #define VERSION "2.1.0"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:3866: error: in `/home/bkerin/tmp/avr-libc-2.1.0':
configure:3868: error: cannot compute suffix of object files: cannot compile
```

dl8dtl, 2022-10-29:
This somehow triggers memories … for whatever reason, your AVR-GCC tries to use the host's (rather than AVR target's) assembler. This is usually an issue with an inconsistent installation between AVR-GCC and AVR-binutils. Make sure they're all using the same `--prefix` when compiling these toolchains.

bkerin, 2022-10-29:
Yes this was the issue.  I had some inconsistently set environment vars following an upgrade, sorry for the bad report.

In process I learned one interesting thing which is that after a successful avr-gcc build the link step of a plain compile fails until avr-libc is also installed (complaining: cannot find -lm: No such file or directory and cannof find -lc: No such file or directory).

Closing now.

closed 2022-10-29 by bkerin