avrdudes/avr-libc #900: Current (head) version generates bigger results then much older (ubuntu) version

void-spark, 2022-10-29:
I wish I could provide a more defined question, but I hit my limits here :)

I'm trying to play with a popular flashlight firmware which is avr based.
The sources, and a good automated build on ubuntu with github actions can be found here: https://github.com/void-spark/anduril2
In the docker and docker_arch subfolders are two build environments for local building, based around docker.

The ubuntu one (which is set up the same as the main/current build):
Uses the current ubuntu/debian avr-gcc and avr-libc.
avr-gcc is avr-gcc (GCC) 5.4.0 (yes, rather old)
avr-libc seems to be a mix of 2.0.0, and a version of the atmel sources that can/could be downloaded (marked 2.0.0+Atmel3.6.2-1.1)

The arch one:
Uses the current arch avr-gcc, and a source build of avr-libc of this github repository.
avr-gcc is avr-gcc (GCC) 12.2.0
avr-libc is built from 55e8cac69935657bcd3e4d938750960c757844c3

The problem is that with arch, so the new avr-gcc and avr-libc, each resulting file is some 20-300 bytes larger consistently.
Quite often that means these do not fit.

My questions are,
- is this expected?
- Is this a unavoidable result from new gcc/libc?
- Is there any flags that should be set to get the original sizes again?

Example build steps, same on either env:
```
avr-gcc -DCONFIGFILE=cfg-sofirn-sp36.h -Wall -g -Os -mmcu=attiny85 -c -std=gnu99 -fgnu89-inline -fwhole-program -DATTINY=85 -I.. -I../.. -I../../.. -fshort-enums -I /include/ -o anduril.o -c anduril.c
avr-gcc -Wall -g -Os -mmcu=attiny85 -mrelax -fgnu89-inline -o anduril.elf anduril.o
avr-objcopy --set-section-flags=.eeprom=alloc,load --change-section-lma .eeprom=0 --no-change-warnings -O ihex --remove-section .fuse anduril.elf anduril.hex
```

Of course I can keep using a old compiler, and cobbled together libc, as they are in debian/ubuntu.
But I'd like to be able to use 2.2.0 when it comes out :)

On the upside, the current main does seem to build everything, including attiny1616 (new chips) :)

dl8dtl, 2022-10-29:
Well, this is not an avr-libc issue, really. So I'm going to mark it as closed.
I've also noticed the recent GCCs are quite poor in their optimization in some situations. Keep in mind that the former backend infrastructure in GCC AVR-GCC was using has been completely abandoned in some recent version (IIRC it was GCC 10), so the current AVR implementation is quite new.
I'm not sure about how many active developers are there now in GCC working on AVR (if any), so I can't tell you how much you might expect from filing "missed optimization" bug reports there. But you could give it a try at least.

closed 2022-10-29 by dl8dtl

void-spark, 2022-10-29:
It's useful information at least, thank you :)