avrdudes/avr-libc #903: alpine packages for `armv7` and `armhf` arch

BuckarooBanzay, 2022-11-27:
(please point me into the right direction if this is the wrong place for this)

The alpine package repo doesn't seem to have the `armv7` and `armhf` architecture
variants of the `avr-libc` package anymore in version 3.17:

https://pkgs.alpinelinux.org/packages?name=avr-libc&branch=v3.17&repo=&arch=&maintainer=

What's the reasoning behind this and how can i help to re-add them?

dl8dtl, 2022-11-27:
I have no ideas about alpine, however, keep in mind that avr-libc is architecture agnostic with respect to the host machine. It provides header files and libraries to be used for the AVR target architecture only, together with AVR-GCC and AVR binutils as a cross-compiler.
So you ought to be able to use the package for *any* host system architecture.

oliv3r, 2023-04-09:
@BuckarooBanzay you best reach out to https://pkgs.alpinelinux.org/packages?name=avr-libc&branch=v3.17&repo=&arch=&maintainer=Stefan%20Wagner who is the maintainer.

However, we can see in this commit, that Psykose disabled armv7 here: https://git.alpinelinux.org/aports/commit/community/avr-libc/APKBUILD?h=3.17-stable&id=0103dcfffb88f5136fec512693399df562933fc1 sadly, no commit entry as to why.

So either reach out to either of them, or open an issue on the alpine bugtracker.

My guess however, is that for some reason it wouldn't compile with the latest compiler, and that would be an issue for this bug tracker. But that's just me guessing :)

I think other then that, it's fair to close this issue, as it's not relevant otherwise.

sprintersb, 2024-07-25:
alpine issue as explained above

closed 2024-07-25 by sprintersb