avrdudes/avr-libc #904: Support for tinyAVR 2 devices

DominikRzecki, 2023-01-08:
Hello,
I would like to know if there will be support for the following devices:
ATtiny424,426,427
ATtiny824,826,827
ATtiny1624,1626,1627
ATtiny3224,3226,3227
Thanks

sprintersb, 2023-03-29:
Supporting "unsupported" devices is usually done by means of `.atpack` files provided by Atmel / Microchip.

Maybe have a look at [Supporting "unsupported" Devices](https://gcc.gnu.org/wiki/avr-gcc#Supporting_.22unsupported.22_Devices) in the avr-gcc wiki, and come back if you have any remaining questions.

sprintersb, 2023-08-04:
FYI, support for these devices has been added to avr-gcc v13.3+, see https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=85414e25ade3db84baeba4acef296e4f34f4184b

The bulk of work for full support of the devices (without using ATPACK) is on the side of AVR-LibC, though.

Apart from that, I'd recomment to use a compiler with reasonable code generation like avr-gcc v8, which implies that you have to resort to ATPACK, of course.

sprintersb, 2024-05-31:
The answer to the top question is: yes. All these devices are supported now, see current NEWS for example.

closed 2024-05-31 by sprintersb