avrdudes/avr-libc #910: Not inverted fuse bits

stefanrueger, 2023-03-22:
Normally fuse bits are defined like
```
#define FUSE_SOMETHING (unsigned char)~_BV(some_n)
```

For the list of parts below gcc v12 defines a lot of them *without bit inversion* `~`
```
atmega1608
atmega1609
atmega3208
atmega3209
atmega4808
atmega4809
atmega808
atmega809
attiny1604
attiny1606
attiny1607
attiny1614
attiny1616
attiny1617
attiny202
attiny204
attiny212
attiny214
attiny3214
attiny3216
attiny3217
attiny402
attiny404
attiny406
attiny412
attiny414
attiny416
attiny417
attiny804
attiny806
attiny807
attiny814
attiny816
attiny817
```

One example is here: https://github.com/avrdudes/avr-libc/blob/55e8cac69935657bcd3e4d938750960c757844c3/include/avr/iotn1616.h#L5613-L5672

stefanrueger, 2023-03-25:
Apparently, "modern" AVRs (ie, the AVR8X family but not XMEGAs) treat some of their fuses as configuration values that are copied into a live register after power-on reset. This makes it more "practical" to treat a bit value 0 as false/disable whilst the tradition used to be that fuse bits with a value of 0 are treated as programmed/enable. 

Fun fact: The EESAVE fuse bit  has the latter meaning in "classic"/XMEGAs, whilst it has the semantically opposite meaning for modern AVRs. And to make it more fun for programmers/users the semantics of the reset pin configuration (acting as GPIO or acting as external reset) has remained the same between classic and modern parts, but the decision was taken to *rename* this AVR fuse bit from `RSTDISBL` to `RSTPINCFG`.

Summarising, sometimes Microchip keep the name of the fuse bit but *invert* the semantics of its value, and sometimes Microchip keep the semantics of the bit value but *change* the name. Go figure!

sprintersb, 2025-09-14:
IMO what we don't want is to introduce incompatibilities by patching the I/O header and increasing the confision even more.

To that end, added a warning in the `avr/fuses.h` documentation.