avrdudes/avr-libc #912: Philosophy of `_bm` vs `_gm` suffixes

stefanrueger, 2023-03-22:
One ostensibly signifies the notion of a one-bit bit-mask while the other denote multi-bit bitmasks. But when is distinction in the constant *name* actually useful? I mean when I don't know whether something (say a clock frequency selector) has two (one bit) or more (multiple bits) selections, how should I proceed?

On the other hand I can always count the number of bits set when I have the pattern. The *name* of the constant does not need to encode that.

Would it not be better to always use the suffix `_bm`?

dl8dtl, 2023-03-22:
IIRC these names came straight from Atmel.

stefanrueger, 2023-03-22:
> IIRC these names came straight from Atmel.

Should employ more people who pass the Sally-Anne-test.

dl8dtl, 2023-03-22:
Btw., I'm always hesitant to change things that have been that way for 10+ years. Regardless of whether they make any sense or not, you can never know who might actually be using them out there.

stefanrueger, 2023-03-22:
Sure, can and should keep all there is, but that wouldn't preclude duplicating those `_gm` constants also as `_bm` constants (or a new one `_mask`). I admit it's probably not worth the trouble, as there are much bigger inconsistencies in naming registers etc - this was actually more of a question out of curiosity (as I didn't want to presume there wasn't much thought behind these decisions).

closed 2025-04-05 by sprintersb