avrdudes/avr-libc #928: _delay_ms() and AtTiny1606

JKBert, 2023-08-14:
Hello,
I open a issue as I have checked my code and I don't understand why _delay_ms(1000) stops CPU for... 3 seconds.
I have built a devel version of avr-libc as I use a Tiny1606 in a new projet. This CPU runs @ 10 Mhz and F_CPU is defined in my firmware.h file:
```
#ifndef __AVR_ATtiny1606__
#   define __AVR_ATtiny1606__
#endif

#ifndef FIRMWARE_H
#   define FIRMWARE_H
#   define F_CPU   10000000UL

#   include <stdio.h>
#   include <stdlib.h>
#   include <avr/io.h>
#   include <avr/interrupt.h>
#   include <util/delay.h>
    
#   ifdef __MAIN__
#       define  _EXTERN_
#       define  _DECL_(v, ...)  v = __VA_ARGS__
#   else
#       define  _EXTERN_    extern
#       define  _DECL_(v, ...)  v
#   endif 
```

In main.c, I have configured master clock:
```
    // 10 MHz
    CLKCTRL.MCLKCTRLA = 0x00;
    CLKCTRL.MCLKCTRLB = CLKCTRL_PDIV_2X_gc | CLKCTRL_PEN_bm;

    // Debug
    while(1)
    {
    for(i = 0x02; i < 0x10; i <<= 1)
    {
        PORTC.OUT = i;
        _delay_ms(1000);
    }
    }
```
Why does _delay_ms() stop CPU for 3 seconds ?
Regards,
JB

apcountryman, 2023-08-14:
The CLKCTRL.MCLKCTRLA and CLKCTRL.MCLKCTRLB registers are under configuration change protection (CCP). To write to a register that is under CCP, you must write the unlock protected I/O registers key to the CPU.CCP register and then write the data to the protected register within four instructions. To execute the unlock sequence and ensure that the timing requirements are met, use `_PROTECTED_WRITE()`.
```c
_PROTECTED_WRITE( CLKCTRL.MCLKCTRLA, 0x00 );
_PROTECTED_WRITE( CLKCTRL.MCLKCTRLB, CLKCTRL_PDIV_2X_gc | CLKCTRL_PEN_bm );
```

apcountryman, 2023-08-14:
> The CLKCTRL.MCLKCTRLA and CLKCTRL.MCLKCTRLB registers are under configuration change protection (CCP). To write to a register that is under CCP, you must write the unlock protected I/O registers key [...]

Also, the code should probably wait for the clock source change to complete before proceeding to the timing check.
```c
_PROTECTED_WRITE( CLKCTRL.MCLKCTRLA, 0x00 );
_PROTECTED_WRITE( CLKCTRL.MCLKCTRLB, CLKCTRL_PDIV_2X_gc | CLKCTRL_PEN_bm );

while ( CLKCTRL.MCLKSTATUS & CLKCTRL_SOSC_bm ) {}
```

JKBert, 2023-08-14:
Thanks. It runs now as expected.
Best regards,
JB