avrdudes/avr-libc #930: Fail to build from sources on RISC-V64 machine

moui0, 2023-10-10:
The config.guess file contained in source tarball is too old that it cannot guess build type on RISC-V64. It can be updated by executing `autoreconf -fiv`.

Full configure error output:
```
checking build system type... ./config.guess: unable to guess system type

This script, last modified 2003-07-02, has failed to recognize
the operating system you are using. It is advised that you
download the most up to date version of the config scripts from

    ftp://ftp.gnu.org/pub/gnu/config/

If the version you run (./config.guess) is already up to date, please
send the following data and any information you think might be
pertinent to <config-patches@gnu.org> in order to provide the needed
information to handle your system.

config.guess timestamp = 2003-07-02

uname -m = riscv64
uname -r = 6.4.10-arch1-1
uname -s = Linux
uname -v = #1 SMP PREEMPT_DYNAMIC Fri, 11 Aug 2023 11:03:36 +0000

/usr/bin/uname -p = unknown
/bin/uname -X     = 

hostinfo               = 
/bin/universe          = 
/usr/bin/arch -k       = 
/bin/arch              = 
/usr/bin/oslevel       = 
/usr/convex/getsysinfo = 

UNAME_MACHINE = riscv64
UNAME_RELEASE = 6.4.10-arch1-1
UNAME_SYSTEM  = Linux
UNAME_VERSION = #1 SMP PREEMPT_DYNAMIC Fri, 11 Aug 2023 11:03:36 +0000
configure: error: cannot guess build type; you must specify one
```

sprintersb, 2024-02-03:
`config.guess` is generated by `bootstrap` which runs autotools to generate it. So it seems there is a problem on your build machine?

kxxt, 2024-07-16:
> `config.guess` is generated by `bootstrap` which runs autotools to generate it. 

This is not true for the bz2 tarball in releases. The bz2 tarball includes `config.guess`, which is copied from the machine that generates the release tarball. 

I checked 2.2.0 release tarball and this issue should be fixed now.

sprintersb, 2024-07-16:
> The bz2 tarball includes `config.guess`

So then it depends on which autotools Jörg (@dl8dtl) is using to roll the release.

dl8dtl, 2024-07-16:
> I checked 2.2.0 release tarball and this issue should be fixed now.

Fine then, I certainly won't downgrade it in future. ;-)

So I guess it can be closed then.

closed 2024-07-16 by dl8dtl