avrdudes/avr-libc #934: EEPROM doesn't get written on AVR64EA32. I expect AVR DD to also be affected...

dragontamer, 2023-12-14:
I'm running Microchip Studio 7.0.132 on Windows 10 and am using avr-libc.

Upon trying to use eeprom_write_dword, the code fails to update the EEPROM. I did a bit of digging in the .lss and have come across the following definition of eeprom_write_r18.

```0000187a <eeprom_write_r18>:
    187a:	af 93       	push	r26
    187c:	bf 93       	push	r27
    187e:	e0 e0       	ldi	r30, 0x00	; 0
    1880:	f0 e1       	ldi	r31, 0x10	; 16
    1882:	36 81       	ldd	r19, Z+6	; 0x06
    1884:	30 fd       	sbrc	r19, 0
    1886:	fd cf       	rjmp	.-6      	; 0x1882 <eeprom_write_r18+0x8>
    1888:	dc 01       	movw	r26, r24
    188a:	a0 50       	subi	r26, 0x00	; 0
    188c:	bc 4e       	sbci	r27, 0xEC	; 236
    188e:	2c 93       	st	X, r18
    1890:	3d e9       	ldi	r19, 0x9D	; 157
    1892:	34 bf       	out	0x34, r19	; 52
    1894:	23 e0       	ldi	r18, 0x03	; 3 <------ Seemingly the issue
    1896:	20 83       	st	Z, r18
    1898:	01 96       	adiw	r24, 0x01	; 1
    189a:	bf 91       	pop	r27
    189c:	af 91       	pop	r26
    189e:	08 95       	ret
```

On address 1894 and 1896, we can see that the value "3" is being written to Z-register (which at this point should be address 0x1000, aka NVMCTRL_CTRLA). 0x3 does not correspond to any value in AVR64EA32 (or AVR DD for that matter). I'm no expert on AVR, but I expect the value EEPERW (0x15 on AVR EA, and 0x13 on AVR DD) to be written here to perform the eeprom-Erase+Write.

3 seems to be complete nonsense, and I have no idea how it got here. There's probably some kind of #define or #if bug in avr-libc that's getting the wrong value of EEPERW somehow.

----------

I can see that in https://github.com/avrdudes/avr-libc/blob/main/libc/misc/eewr_byte.S , the lines in question relate to:

```
	ldi	r18, CCP_SPM_gc
	out CCP, r18
	ldi	r18, NVMCTRL_CMD_PAGEERASEWRITE_gc
	std	Z + NVMCTRL_CTRLA - NVM_BASE, r18
```

So NVMCTRL_CMD_PAGEERASEWRITE_gc "should" equal NVMCTRL_CMD_EEPERW_gc.

-----------

Seemingly, I can add "	_PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_EEPERW_gc);" and my code starts to work, so the ioavr64ea32.h file is correct in my configuration. Its just not mapping to "NVMCTRL_CMD_PAGEERASEWRITE_gc" correctly somehow.

dragontamer, 2023-12-14:
Apparently I was on "Initial Release" 1.0.38 device support. Woops.

Well I feel like I wasted a bit of my own time tracking this down. On the newest 2.8.189 device support, this issue is fixed. It doesn't seem documented anywhere but yeah, if anyone else has this issue make sure to update to the latest device support package.

sprintersb, 2024-02-17:
Should be fixed by commit https://github.com/avrdudes/avr-libc/commit/f8627a2a99905e7db68c1a96d3d4bd6cfba7385f

closed 2024-02-17 by sprintersb