avrdudes/avr-libc #944: Many headers use // comments, hence don't work with pre-C99

sprintersb, 2024-02-09:
Many of the device headers have `//` comments that only work with C++ or from  C99 on. Searching for `//` comments with, say

`find . -name '*.h' -exec grep -H // {} ';'`

finds 400 matches.

dl8dtl, 2024-02-09:
I'm not so sure whether it's necessary to support pre-C99. Also, various compilers used to support // comments already before the C99 release as an extension.

IIRC, we also use named struct initialization in header files, which is also a C99 feature.

sprintersb, 2024-02-09:
> we also use named struct initialization in header files, which is also a C99 feature.

Which won't work with C++ older than C++20, see #898 for an example.

Closing this one.

closed 2024-02-09 by sprintersb

dl8dtl, 2024-02-09:
> > we also use named struct initialization in header files, which is also a C99 feature.
> 
> Which won't work with C++ older than C++20

Alas, yes, I know.

I'd probably have preferred a C++ constructor instead of falling back to pre-C99 …