avrdudes/avr-libc #950: Current web documentation has trashed code indentations

sprintersb, 2024-03-17:
Due to a Doxygen bug introduced in v1.8.1.2, the indentations in `\code` regions are gone.

See for example the online [FAQ](https://www.nongnu.org/avr-libc/user-manual/FAQ.html#faq_flashstrings) which is generated with Doxygen v1.8.7 (as of 2024-03-17).

See also [stackoverflow: Doxygen: `\code` tag kills indentation](https://stackoverflow.com/questions/11190838/doxygen-code-tag-kills-indentation).  A comment says that the bug is fixed in v1.8.7 (which it obviously is not...)

My local Doxygen version is v1.9.1 and the indentations work as expected.

dl8dtl, 2024-03-17:
My local version is 1.9.6. Guess I could try rebuilding everything …

sprintersb, 2024-03-24:
Re-building would also fix #597. (Though https://www.nongnu.org/avr-libc/user-manual/using_tools.html is pretty much outdated).

sprintersb, 2024-06-07:
Fixed by using an updated Doxygen version (v1.9.1).

closed 2024-06-07 by sprintersb