avrdudes/avr-libc #953: [Documentation] break and continue statements inside ATOMIC_BLOCK

YBerd, 2024-03-24:
Hello here,
There is a non-obvious ATOMIC_BLOCK behavior that is not covered in docs.

ATOMIC_BLOCK is implemented as for() statement, therefore break and continue inside the block will exit the block, not the surrounding loop.

For example

```
while(...)
{
    if (someTest(someVolatileGlobalThing))
        break;
        
    .....
}
```

works just fine. But a naive attempt to make the check safe

```
while(...)
{
    ATOMIC_BLOCK(ATOMIC_FORCEON)
    {
        if (someTest(someVolatileGlobalThing))
            break;
    }
    
    .....
}
```
creates an endless loop that is hard to spot.

Workaround could be like this

```
while(...)
{
    bool flag;
    
    ATOMIC_BLOCK(ATOMIC_FORCEON)
    {
        flag = someTest(someVolatileGlobalThing);
    }
    
    if (flag)
        break;
    
    .....
}
```

This is not a big issue overall, but is worth mentioning in ATOMIC_BLOCK documentation.

sprintersb, 2024-03-24:
Added warning https://github.com/avrdudes/avr-libc/commit/1c5c4586e1b9ef7395063cb7f9cde64f7507e350 to the documentation.  Hope that is clear enough?

closed 2024-03-24 by sprintersb

YBerd, 2024-03-24:
Looks good, thanks for the fast response!