avrdudes/avr-libc #955: avr-libc/include/avr/version.h.in

paulw517, 2024-04-03:
When compiling an Arduino sketch, it returns an error <arv/version.h> no file or folder found

the arb-libc/include/avr library lists version.h.in

Is this correct?

dl8dtl, 2024-04-03:
Whoever has packaged that avr-libc distribution must have done their job wrong.
`avr/version.h.in` is the template, it is supposed to be "personalized" during the build process, resulting in `avr/version.h`.
For example, my `avr/version.h` here looks like (comment from top omitted for brevity):
```
#ifndef _AVR_VERSION_H_
#define _AVR_VERSION_H_

/** \ingroup avr_version
    String literal representation of the current library version. */
#define __AVR_LIBC_VERSION_STRING__ "2.2.0git"

/** \ingroup avr_version
    Numerical representation of the current library version.

    In the numerical representation, the major number is multiplied by
    10000, the minor number by 100, and all three parts are then
    added.  It is intented to provide a monotonically increasing
    numerical value that can easily be used in numerical checks.
 */
#define __AVR_LIBC_VERSION__        20200UL

/** \ingroup avr_version
    String literal representation of the release date. */
#define __AVR_LIBC_DATE_STRING__    "20220701"

/** \ingroup avr_version
    Numerical representation of the release date. */
#define __AVR_LIBC_DATE_            20220701UL

/** \ingroup avr_version
    Library major version number. */
#define __AVR_LIBC_MAJOR__          2

/** \ingroup avr_version
    Library minor version number. */
#define __AVR_LIBC_MINOR__          2

/** \ingroup avr_version
    Library revision number. */
#define __AVR_LIBC_REVISION__       0

#endif /* _AVR_VERSION_H_ */
```

paulw517, 2024-04-03:
Thank you for you response dI8dtI, what should be done to rectify this issue?

dl8dtl, 2024-04-03:
Please ask whoever packaged your binary version of avr-libc.

You mention Arduino, is it their version?

paulw517, 2024-04-03:
I have no idea who packaged it; I was simply using a download of the code directly from GitHub to add to my library to refernce in my Arduino sketch?

paulw517, 2024-04-03:
Thank you dI8dtI for your response, what should be done to rectify this issue?

Paul ***@***.*** the heart of the beautiful Manistee National Forest

sprintersb, 2024-04-12:
> I was simply using a download of the code directly from GitHub to add to my library to refernce in my Arduino sketch?

That won't work. You'll have to do a proper build and installation of the library.  As you are likely using a pre-installed toolchain, building and installing a new libc version means to install into an already existing installation which is strongly discouraged.

Hence, when you really want an updated AVR-LibC revision, the path to go is to configure-build-install Binutils, GCC and AVR-LibC, preferably into a location that does not require root access.

Presumably you just want to add support for some new devices.  To that end, use [atpack](https://gcc.gnu.org/wiki/avr-gcc#atpack) files as distributed by the hardware manufacturer.

sprintersb, 2024-04-12:
Closed. Using blobs from GitHub won't work; the reporter did not install / bundle the software, which won't work.