avrdudes/avr-libc #958: NVM_PROD_SIGNATURES_t is missing from iox128a1u.h

lmorrisonnautel, 2024-04-23:
The production signature row structure is defined on a per-model basis in most iox*.h headers, but it's missing for ATxmega128A1U.

It should follow largely the same structure as its older sibling the ATxmega128A1, but with some additions for its extra USB and DAC related features:
```
typedef struct NVM_PROD_SIGNATURES_struct
{
    register8_t RCOSC2M;  /* RCOSC 2MHz Calibration Value B */
    register8_t RCOSC2MA; /* RCOSC 2MHz Calibration Value A */
    register8_t RCOSC32K;  /* RCOSC 32kHz Calibration Value */
    register8_t RCOSC32M;  /* RCOSC 32MHz Calibration Value B */
    register8_t RCOSC32MA; /* RCOSC 32MHz Calibration Value A */
    register8_t reserved_0x05;
    register8_t reserved_0x06;
    register8_t reserved_0x07;
    register8_t LOTNUM0;  /* Lot Number Byte 0, ASCII */
    register8_t LOTNUM1;  /* Lot Number Byte 1, ASCII */
    register8_t LOTNUM2;  /* Lot Number Byte 2, ASCII */
    register8_t LOTNUM3;  /* Lot Number Byte 3, ASCII */
    register8_t LOTNUM4;  /* Lot Number Byte 4, ASCII */
    register8_t LOTNUM5;  /* Lot Number Byte 5, ASCII */
    register8_t reserved_0x0E;
    register8_t reserved_0x0F;
    register8_t WAFNUM;  /* Wafer Number */
    register8_t reserved_0x11;
    register8_t COORDX0;  /* Wafer Coordinate X Byte 0 */
    register8_t COORDX1;  /* Wafer Coordinate X Byte 1 */
    register8_t COORDY0;  /* Wafer Coordinate Y Byte 0 */
    register8_t COORDY1;  /* Wafer Coordinate Y Byte 1 */
    register8_t reserved_0x16;
    register8_t reserved_0x17;
    register8_t reserved_0x18;
    register8_t reserved_0x19;
    register8_t USBCAL0;  /* USB Pad Calibration Byte 0 */
    register8_t USBCAL1;  /* USB Pad Calibration Byte 1 */
    register8_t RCOSC48M;  /* 48MHz RSCOSC Calibration (overclocks the 32MHZ RCOSC for use with USB) */
    register8_t reserved_0x1D;
    register8_t reserved_0x1E;
    register8_t reserved_0x1F;
    register8_t ADCACAL0;  /* ADCA Calibration Byte 0 */
    register8_t ADCACAL1;  /* ADCA Calibration Byte 1 */
    register8_t reserved_0x22;
    register8_t reserved_0x23;
    register8_t ADCBCAL0;  /* ADCB Calibration Byte 0 */
    register8_t ADCBCAL1;  /* ADCB Calibration Byte 1 */
    register8_t reserved_0x26;
    register8_t reserved_0x27;
    register8_t reserved_0x28;
    register8_t reserved_0x29;
    register8_t reserved_0x2A;
    register8_t reserved_0x2B;
    register8_t reserved_0x2C;
    register8_t reserved_0x2D;
    register8_t TEMPSENSE0;  /* Temperature Sensor Calibration Byte 0 */
    register8_t TEMPSENSE1;  /* Temperature Sensor Calibration Byte 0 */
    register8_t DACA0OFFCAL;  /* DACA0 Calibration Byte 0 */
    register8_t DACA0GAINCAL;  /* DACA0 Calibration Byte 1 */
    register8_t DACB0OFFCAL;  /* DACB0 Calibration Byte 0 */
    register8_t DACB0GAINCAL;  /* DACB0 Calibration Byte 1 */
    register8_t DACA1OFFCAL;  /* DACA1 Calibration Byte 0 */
    register8_t DACA1GAINCAL;  /* DACA1 Calibration Byte 1 */
    register8_t DACB1OFFCAL;  /* DACB1 Calibration Byte 0 */
    register8_t DACB1GAINCAL;  /* DACB1 Calibration Byte 1 */
    register8_t reserved_0x38;
    register8_t reserved_0x39;
    register8_t reserved_0x3A;
    register8_t reserved_0x3B;
    register8_t reserved_0x3C;
    register8_t reserved_0x3D;
    register8_t reserved_0x3E;
} NVM_PROD_SIGNATURES_t;
```

sprintersb, 2024-04-25:
Would https://github.com/sprintersb/avr-libc/commit/fca9aedb8e20cf506d3524f9ea1ca1a6dd35772c fix the issue?

lmorrisonnautel, 2024-04-25:
Yes I believe it would.

sprintersb, 2024-04-25:
Fixed by https://github.com/avrdudes/avr-libc/commit/4817a64b871acd73c3b8916f81feee6a4e039a5e.

closed 2024-04-25 by sprintersb