avrdudes/avr-libc #965: GCC test gcc.target/avr/torture/addr-space-2-1.c started failing

dinuxbg, 2024-06-02:
A few GCC tests started failing with the following error:
```
avr/bin/ld: section .progmemx.data.ftoa_engine LMA [00010023,0001011c] overlaps section .data LMA [00010023,0001003e]
FAIL: gcc.target/avr/torture/addr-space-2-1.c   -O0  (test for excess errors)
```

Affected tests:
 -  `gcc.target/avr/torture/builtins-2-flash.c`
 - `gcc.target/avr/torture/addr-space-2-1.c `
 - `gcc.target/avr/torture/addr-space-1-1.c`

It probably started with avr-libc commit b74dd415579f4c68d57f9795650800242785974f

Last known working toolchain:
```
binutils 84f7f8940760d0eaeb51a38b345db369b4207e5b
gcc a6114c2a691112f9cf5b072c21685d2e43c76d81
avrlibc aa72d2d0020cbca603baeccf7fa6956d404ce17b
winavr a754eeff1a7ba37063114e79a83f72a9a277e157
```

First toolchain to fail `gcc.target/avr/torture/addr-space-2-1.c`:
```
binutils 89bd22ef5b7fec9ac9e760de276781b181f58ee0
gcc 3c7301f947ec27e5427106c8fb053693b1c854af
avrlibc 32ffbdbae34d84b9dfd0c3ff8f7dc38c1dc269f6
winavr a754eeff1a7ba37063114e79a83f72a9a277e157
```

The script for testing avr-gcc I'm using is https://github.com/dinuxbg/gnupru/blob/master/testing/buildbot-avr.sh

sprintersb, 2024-06-02:
This is an avrtest issue.  Some tests use the ` avr51-flash1.x` ld script from avrtest which missed the `.progmemx.*` input sections (they are present in the default ld scripts), hence `.progmemx.*` where orphans.

I updated avrtest. For a detailed description, see [avrtest::BUGS](https://github.com/sprintersb/atest/blob/master/BUGS).

FYI, `.progmemx` is at a high address since https://sourceware.org/PR21849. Prior to that, it was at a low address due to `.progmem*`.

See also issue #962 which now uses `.progmemx.*` throughout AVR-LibC.

dinuxbg, 2024-06-02:
Thank you. I tested the following versions, and indeed latest atest fixes the issue:
```
binutils 89bd22ef5b7fec9ac9e760de276781b181f58ee0
gcc 80c1a963bdfb573feb398ef02269671a5d690c1f
avrlibc 32ffbdbae34d84b9dfd0c3ff8f7dc38c1dc269f6
atest cefa8f0407b25a57b6afcc1076fdc453961cb640
```

closed 2024-06-02 by dinuxbg