avrdudes/avr-libc #966: LaTeX documentation generation is broken

dl8dtl, 2024-06-02:
For some reasons, the LaTeX documentation no longer works. It used to work completely up to release 2.0, and then subsequently broke.

This is a placeholder to remind it still needs fixing, even though release 2.2.0 might ship with a semi-handcrafted PDF version.

dl8dtl, 2024-06-02:
Update: it's the table "Output Sections and Memory Regions" in `sections.dox` that causes the pdflatex to fail.

sprintersb, 2024-06-07:
The changes are integrated now, most notably b749433463cb7010ecf7ed8e98f40c249c0e1b56, b0c0f4c2d75183c049dc32dcaf65173806bc0d36, a348287a985104aa2e27afff4076a9fa2e8dab31, fc692172790a235e749750d89d2993e5900f331f, a7ba0c1f93caabf58ef87ca34aa87f1bccac7f05, 6b6a8103b241c591bd47cfcdf6dd8d6b603dae37.

Also the used Doxygen version is crucial, IIUC we are using **Doxygen v1.9.1** now.  I tried v1.9.8 which does not work, same with v1.12 from Doxygen master, and Jörg has tried v1.11 with no avail. So many issues there...

sprintersb, 2024-07-18:
"Solved" I guess, by using *exactly* Doxygen v1.9.6.

dl8dtl, 2024-07-19:
Unfortunately, seems we by now require this version.

closed 2024-07-19 by dl8dtl