avrdudes/avr-libc #976: Documentation outdated on instructions how to install

eiaro, 2024-08-05:
Quote from [Building and Installing the GNU Tool Chain](https://avrdudes.github.io/avr-libc/avr-libc-user-manual-2.2.0/install_tools.html#install_avr_libc)
>To build and install AVR-LibC:
>
>```bash
>$ # in ./source
>$ tar xfz avr-libc-<version>.tar.gz
>
>$ # in ./build
>$ mkdir avr-libc-<version>
>$ cd avr-libc-<version>
>$ ../../source/avr-libc-<version>/configure --prefix=$PREFIX \
>    --build=x86_64-pc-linux-gnu --host=avr
>$ make
>$ make install
>```

As there isn't a `configure` file to run, this makes it difficult to follow the instructions.

The file exist on the 2.1.0 release from Savannah, but is not in the git repo on GitHub with 2.1 tag

sprintersb, 2024-08-05:
> The file [configure] exist on the 2.1.0 release from Savannah, but is not in the git repo on GitHub with 2.1 tag

Two lines above your quotation, there is a fat note that says:

> **Note:** If you have obtained the latest AVR-LibC from [git](https://avrdudes.github.io/avr-libc/avr-libc-user-manual-2.2.0/install_tools.html#using_git), you will have to run the `./bootstrap` script before using either of the build methods described below.

So did you run `./bootstrap` ?

Or are you saying that `configure` is missing from the v2.2.0 release ball?

eiaro, 2024-08-05:
The file is included in the .bz2, but not in .gz or .zip. So the clause on 'where' you obtain the library is irrelevant. So is the 'latest' as the same issue will be for older releases as well.

Maybe something like this:
> **Note:** Depending on how you obtained AVR LibC, you may need to run the `./bootstrap` script....

closed 2024-08-05 by eiaro