avrdudes/avr-libc #987: Unit test for `realloc` fails due to GCC 14.2.0 optimizations

sciprosk, 2024-09-22:
I am using Binutils 2.43.1 and GCC 14.2 to compile and run unit tests in the GitHub library repo. I see several tests fail related to `malloc`/`realloc` test code.  Below, I investigate the case of `realloc-3.c` test in the `stdlib` test suite. The test fails only with enabled compiler optimizations. This failure is *not* due to the code rearrangements, but rather due additional library code  injected into `__init`.

### Steps to reproduce
```
$ cd tests/simulate
```
```
$ avr-gcc -Wundef -I. -gdwarf-4 -W -Wall -Os -pipe -Wno-array-bounds -I../../include -I ../../include -nostdlib -std=gnu99 -mmcu=atmega128 ../../avr/devices/atmega128/crtatmega128.o stdlib/realloc-3.c ../../avr/lib/avr51/libc.a ../../avr/lib/avr51/libm.a ../../avr/devices/atmega128/libatmega128.a -lgcc -o realloc-3.elf
```
```
$ simulavr --version
SimulAVR 1.1.0
```
```
$ simulavr -d atmega128 -T __stop_program -f realloc-3.elf -C core_avr_dump.core -m 60000000000

FATAL: file /home/buildbudy/simulavr/libsim/decoder.cpp: line 1777: Illegal opcode 'ff ff' executed at PC=0x14af0 (42360)! Simulation terminated!
```
### Disassembling
Using `avr-gdb` from Binutils 2.43.1, shows that the problems is in the `__init` code in the function
``` C
void fill_mem(void) __attribute__((section(".init3"),naked));
void fill_mem(void)
{
  extern uint8_t __bss_end;
  uint8_t *p = &__bss_end;
  do
    *p++ = 0xa5;
  while (p < (uint8_t *)RAMEND);
}
```
Optimized build calls `memset` at `fill_mem+30`. Note that there is no such call when compiler optimizations are disabled.
``` Assembler
Dump of assembler code from 0xc2 to 0x126:
   0x000000c2 <__init+0>:       eor     r1, r1
   0x000000c4 <__init+2>:       out     0x3f, r1        ; 63
   0x000000c6 <__init+4>:       ldi     r28, 0xFF       ; 255
   0x000000c8 <__init+6>:       ldi     r29, 0x10       ; 16
   0x000000ca <__init+8>:       out     0x3e, r29       ; 62
   0x000000cc <__init+10>:      out     0x3d, r28       ; 61
   0x000000ce <fill_mem+0>:     ldi     r24, 0x2C       ; 44
   0x000000d0 <fill_mem+2>:     ldi     r25, 0x01       ; 1
   0x000000d2 <fill_mem+4>:     ldi     r20, 0xFF       ; 255
   0x000000d4 <fill_mem+6>:     ldi     r21, 0x10       ; 16
   0x000000d6 <fill_mem+8>:     subi    r20, 0x2B       ; 43
   0x000000d8 <fill_mem+10>:    sbci    r21, 0x01       ; 1
   0x000000da <fill_mem+12>:    cp      r24, r1
   0x000000dc <fill_mem+14>:    sbci    r25, 0x11       ; 17
   0x000000de <fill_mem+16>:    brcs    .+4             ;  0xe4 <fill_mem+22>
   0x000000e0 <fill_mem+18>:    ldi     r20, 0x01       ; 1
   0x000000e2 <fill_mem+20>:    ldi     r21, 0x00       ; 0
   0x000000e4 <fill_mem+22>:    ldi     r22, 0xA5       ; 165
   0x000000e6 <fill_mem+24>:    ldi     r23, 0x00       ; 0
   0x000000e8 <fill_mem+26>:    ldi     r24, 0x2B       ; 43
   0x000000ea <fill_mem+28>:    ldi     r25, 0x01       ; 1
=> 0x000000ec <fill_mem+30>:    call    0x97e   ;  0x97e <memset>
   0x000000f0 <__do_copy_data+0>:       ldi     r17, 0x01       ; 1
   0x000000f2 <__do_copy_data+2>:       ldi     r26, 0x00       ; 0
   0x000000f4 <__do_copy_data+4>:       ldi     r27, 0x01       ; 1
   0x000000f6 <__do_copy_data+6>:       ldi     r30, 0x56       ; 86
   0x000000f8 <__do_copy_data+8>:       ldi     r31, 0x0A       ; 10
   0x000000fa <__do_copy_data+10>:      ldi     r16, 0x00       ; 0
   0x000000fc <__do_copy_data+12>:      out     0x3b, r16       ; 59
   0x000000fe <__do_copy_data+14>:      rjmp    .+4             ;  0x104 <__do_copy_data+20>
   0x00000100 <__do_copy_data+16>:      elpm    r0, Z+
   0x00000102 <__do_copy_data+18>:      st      X+, r0
   0x00000104 <__do_copy_data+20>:      cpi     r26, 0x1A       ; 26
   0x00000106 <__do_copy_data+22>:      cpc     r27, r17
   0x00000108 <__do_copy_data+24>:      brne    .-10            ;  0x100 <__do_copy_data+16>
   0x0000010a <__do_clear_bss+0>:       ldi     r18, 0x01       ; 1
[... 88 lines cut ...]
   0x00000106 <__do_clear_bss+0>:       ldi     r18, 0x01       ; 1
   0x00000108 <__do_clear_bss+2>:       ldi     r26, 0x1A       ; 26
   0x0000010a <__do_clear_bss+4>:       ldi     r27, 0x01       ; 1
   0x0000010c <__do_clear_bss+6>:       rjmp    .+2             ;  0x110 <__do_clear_bss+10>
   0x0000010e <__do_clear_bss+8>:       st      X+, r1
   0x00000110 <__do_clear_bss+10>:      cpi     r26, 0x2B       ; 43
   0x00000112 <__do_clear_bss+12>:      cpc     r27, r18
   0x00000114 <__do_clear_bss+14>:      brne    .-8             ;  0x10e <__do_clear_bss+8>
   0x00000116 <L0+0>:   call    0x4b2   ;  0x4b2 <main>
```

### Possible solutions?
Changing the code of the `fill_mem` so that it does not write to the end of RAM allows this test to pass with enabled optimizations.
``` C
void fill_mem(void) __attribute__((section(".init3"),naked));
void fill_mem(void)
{
  extern uint8_t __bss_end;
  uint8_t *p = &__bss_end;
  do
    *p++ = 0xa5;
  while (p < (uint8_t *)(RAMEND - 4));
}
```
However, I am not sure if that is a correct solution. The problem is  how GCC makes its optimizations, not the test code.

sprintersb, 2024-09-25:
The option to avoid such optimizations is `-fno-tree-loop-distribute-patterns` (ya, terrible naming and documentation).

Maybe the simplest means to defeat `memset` is `volatile`, like in:

```c++
void fill_mem (void)
{
  extern uint8_t __bss_end;
  uint8_t volatile *p = &__bss_end;
  do
    *p++ = 0xa5;
  while (p < (uint8_t*) (RAMEND - 4));
}
```

sprintersb, 2024-09-25:
Fixed by 2a054cd5c7ed6ad3e6ea87c44655e6309a2d0828.