avrdudes/avr-libc #988: Some tests are failing with SimulAVR but not with AVRtest

sprintersb, 2024-09-28:
Running in `$srcdir/tests/simulate` using SimulAVR (runtest.sh), there are some tests failing:
```none
$ ./runtest.sh -a $builddir -g avr-gcc-15 stdlib/realloc-3.c stdlib/malloc-7.c 
Simulate: stdlib/realloc-3.c atmega128 ... *** simulate failed: 2
Simulate: stdlib/realloc-3.c at90s8515 ... *** simulate failed: 2
Simulate: stdlib/malloc-7.c atmega128 ... *** simulate failed: 103
Simulate: stdlib/malloc-7.c at90s8515 ... *** simulate failed: 103
```
The same tests are passing fine when executed with [AVRtest](https://github.com/sprintersb/atest) (`run-avrtest.sh`):
```none
$ MCUS="atmega128 at90s8515" ./run-avrtest.sh -a $builddir -g avr-gcc-15 stdlib/realloc-3.c stdlib/malloc-7.c 
Using avrtest from: /home/john/gnu/source/avrtest
Simulate avrtest: stdlib/realloc-3.c atmega128 ... OK
Simulate avrtest: stdlib/realloc-3.c at90s8515 ... OK
Simulate avrtest: stdlib/malloc-7.c atmega128 ... OK
Simulate avrtest: stdlib/malloc-7.c at90s8515 ... OK
```
One difference between `runtest.sh` and `run-avrtest.sh` is that the latter assigns more memory (RAM), more that the respective MCU hardware actually supports.  Though I didn't investigate whether that's a factor in this issue.

sciprosk, 2024-09-29:
Same results in SimulAVR with GCC 14.2.0. I looked into the first case today with `realloc-3.c`, and the issue is with trying to dereference `__flp == NULL`. This happens close the end of the simulation when there is only top-most node in the free-list which is eliminated  and set `__flp == NULL` as defined in `malloc.c`:
``` C
       /*
	 * If there's a new topmost chunk, lower __brkval instead.
	 */
	for (fp1 = __flp, fp2 = 0;
	     fp1->nx != 0;
	     fp2 = fp1, fp1 = fp1->nx)
		/* advance to entry just before end of list */;
	cp2 = (char *)&(fp1->nx);
	if (cp2 + fp1->sz == __brkval) {
		if (fp2 == NULL)
			/* Freelist is empty now. */
			__flp = NULL;
		else
			fp2->nx = NULL;
		__brkval = cp2 - sizeof(size_t);
	}
```
This situation is not handled in the test code however. A simple fixture could be to add a null-check.

sprintersb, 2024-09-29:
Fixed by https://github.com/avrdudes/avr-libc/commit/987de777ec6da945d241ad565f99d7ee7c805f53 (for realloc-3.c) and https://github.com/avrdudes/avr-libc/commit/0f1437c381a5f0d627ad5d569e27ced4ad52c39b (for malloc-7.c).

closed 2024-09-29 by sprintersb