avrdudes/avr-libc #990: Programm size regression after commit 6009ae60f0b46aef601f4b8586a51240966dee53

iu1j4, 2024-10-07:
Hello,
commit 6009ae60f0b46aef601f4b8586a51240966dee53 incresed the program size and now it doesnt fit into my atmega1284.
Befor commit 6009ae60f0b46aef601f4b8586a51240966dee53 (Fri Mar 15 17:57:02 2024 +0100 c3c85232b99cd991673928b64e029f80749b8f12) total test programm size is 121568 and it fit into my atmega1284.
After commit 6009ae60f0b46aef601f4b8586a51240966dee53 total test programm size is 123288 and it doesnt fit into my atmega1284 (I use bootloader).
Total size incrase is 1720 bytes since commit 6009ae60f0b46aef601f4b8586a51240966dee53
Can we go back to the crc16.h from commit c3c85232b99cd991673928b64e029f80749b8f12?
thanks for help

sprintersb, 2024-10-07:
So IIUC, then when you are using
```c++
#define __ATTR_ALWAYS_INLINE__ __inline__
#include <util/crc16.h>
// other includes
...
```
then the code size does decrease by over 1 KiB?
Do you have some test case to reproduce?

iu1j4, 2024-10-07:
My fault - I didnt noticed that util/crc16.h contains static inline functions _crc16_update and _crc_ibutton_update that I called directly in my code from few places.
I fixed it by writing my own wrapper function: mycrc16_update() and marked it as __attribute__ ((noinline))

sprintersb, 2024-10-08:
.

sprintersb, 2024-10-08:
.

closed 2024-10-08 by sprintersb