avrdudes/avr-libc #995: How is `crtatmega32u4.o` built from source files?

Semnodime, 2024-10-17:


dl8dtl, 2024-10-18:
You could find this out very simple yourself. Just compile, and log the output:
```
make > logfile.txt 2>&1
```
Then, in `logfile.txt`, search for "atmega32u4".
This is the relevant part:
```
Making all in atmega32u4
gmake[4]: Entering directory '/home/joerg/src/avr-libc/avr/devices/atmega32u4'
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -Wall -W -Wstrict-prototypes -mmcu=atmega32u4  -mcall-prologues -Os  -MT eewr_block_xmega.o -MD -MP -MF .deps/eewr_block_xmega.Tpo -c -o eewr_block_xmega.o ../../../libc/misc/eewr_block_xmega.c
mv -f .deps/eewr_block_xmega.Tpo .deps/eewr_block_xmega.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eerd_block.o -MD -MP -MF .deps/eerd_block.Tpo -c -o eerd_block.o ../../../libc/misc/eerd_block.S
mv -f .deps/eerd_block.Tpo .deps/eerd_block.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eerd_byte.o -MD -MP -MF .deps/eerd_byte.Tpo -c -o eerd_byte.o ../../../libc/misc/eerd_byte.S
mv -f .deps/eerd_byte.Tpo .deps/eerd_byte.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eerd_qword.o -MD -MP -MF .deps/eerd_qword.Tpo -c -o eerd_qword.o ../../../libc/misc/eerd_qword.S
mv -f .deps/eerd_qword.Tpo .deps/eerd_qword.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eerd_dword.o -MD -MP -MF .deps/eerd_dword.Tpo -c -o eerd_dword.o ../../../libc/misc/eerd_dword.S
mv -f .deps/eerd_dword.Tpo .deps/eerd_dword.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eerd_word.o -MD -MP -MF .deps/eerd_word.Tpo -c -o eerd_word.o ../../../libc/misc/eerd_word.S
mv -f .deps/eerd_word.Tpo .deps/eerd_word.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eeupd_block.o -MD -MP -MF .deps/eeupd_block.Tpo -c -o eeupd_block.o ../../../libc/misc/eeupd_block.S
mv -f .deps/eeupd_block.Tpo .deps/eeupd_block.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eeupd_byte.o -MD -MP -MF .deps/eeupd_byte.Tpo -c -o eeupd_byte.o ../../../libc/misc/eeupd_byte.S
mv -f .deps/eeupd_byte.Tpo .deps/eeupd_byte.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eeupd_qword.o -MD -MP -MF .deps/eeupd_qword.Tpo -c -o eeupd_qword.o ../../../libc/misc/eeupd_qword.S
mv -f .deps/eeupd_qword.Tpo .deps/eeupd_qword.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eeupd_dword.o -MD -MP -MF .deps/eeupd_dword.Tpo -c -o eeupd_dword.o ../../../libc/misc/eeupd_dword.S
mv -f .deps/eeupd_dword.Tpo .deps/eeupd_dword.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eeupd_word.o -MD -MP -MF .deps/eeupd_word.Tpo -c -o eeupd_word.o ../../../libc/misc/eeupd_word.S
mv -f .deps/eeupd_word.Tpo .deps/eeupd_word.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eewr_block.o -MD -MP -MF .deps/eewr_block.Tpo -c -o eewr_block.o ../../../libc/misc/eewr_block.S
mv -f .deps/eewr_block.Tpo .deps/eewr_block.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eewr_byte.o -MD -MP -MF .deps/eewr_byte.Tpo -c -o eewr_byte.o ../../../libc/misc/eewr_byte.S
mv -f .deps/eewr_byte.Tpo .deps/eewr_byte.Po
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT eewr_qword.o -MD -MP -MF .deps/eewr_qword.Tpo -c -o eewr_qword.o ../../ [... cut ...]
avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=atmega32u4  -DIOSYMFILE=\"iosym/atmega32u4.S\"   -MT gcrt1.o -MD -MP -MF .deps/gcrt1.Tpo -c -o gcrt1.o ../../../crt1/gcrt1.S
mv -f .deps/gcrt1.Tpo .deps/gcrt1.Po
rm -f libcrt.a
avr-ar cr libcrt.a gcrt1.o 
avr-ranlib libcrt.a
rm -f crtatmega32u4.o
ln gcrt1.o crtatmega32u4.o || cp gcrt1.o crtatmega32u4.o
gmake[4]: Leaving directory '/home/joerg/src/avr-libc/avr/devices/atmega32u4'
```

sprintersb, 2024-10-18:
@dl8dtl  Can you explain what `libcrt.a` is needed for?

Why isn't it enough to just compile `gcrt1.S` to `gcrt1.o` and then movong that file to `crtatmega32u4.o`? Or to simply compile `gcrt1.S` to `crtatmega32u4.o`?

dl8dtl, 2024-10-18:
> @dl8dtl Can you explain what `libcrt.a` is needed for?

No idea, sorry.

closed 2025-04-05 by sprintersb