avrdudes/avr-libc #998: Compatibility Issue with Inline Assembly Constraints in Clang

dakkshesh07, 2024-11-04:
When compiling AVR programs using avr-libc with Clang, an error occurs in header file `avr/wdt.h`

> avr/include/avr/wdt.h:419:12: error: value '64' out of range for constraint 'I'
> : "I" (_SFR_IO_ADDR(_WD_CONTROL_REG)),
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> avr/include/avr/sfr_defs.h:183:27: note: expanded from macro '_SFR_IO_ADDR'
> #define _SFR_IO_ADDR(sfr) (_SFR_MEM_ADDR(sfr) - __SFR_OFFSET)

An initial issue #678 was raised, and it was suggested that the problem stemmed from Clang’s behavior. Specifically, it was noted that `__builtin_constant_p` should cause if (0) branches to be skipped, yet Clang still evaluates constraints within these branches. This reasoning led to the issue being closed as a Clang-related problem.

A related issue https://github.com/llvm/llvm-project/issues/51513 was then opened in the LLVM repository, where it was noted that:

1. **Clang Behavior:** Clang checks inline assembly constraints on operands before eliminating dead branches, causing constraints to be enforced even in branches that would otherwise be removed.

2. **GCC Comparison:** GCC eliminates dead branches before validating constraints. As a result, the `"I"` constraint on `_SFR_IO_ADDR(_WD_CONTROL_REG)` doesn’t trigger an error in GCC, which allows the code to compile successfully.

It was concluded that the fix for this issue should come from the avr-libc side, as the current implementation relies on behavior specific to GCC’s handling of inline assembly constraints, which is fundamentally inconsistent with standard practices for dead code elimination.

dakkshesh07, 2024-11-04:
I found a potential fix that seems to address the constraint validation issues in Clang, but I'm not sure if it’s the right solution. I encourage you to take a look at the commit here: [maribu/avr-libc@2020848](https://github.com/maribu/avr-libc/commit/2020848368efa4ce91c75a0ed9892a8f3418d375).

sprintersb, 2024-11-04:
That "solution" just nullifies the purpose of the `I` constraint, which is to diagnose early (already in the compiler and not in the assembler) when a const operand doesn't fit.

sprintersb, 2025-04-17:
clang issue

closed 2025-04-17 by sprintersb

benshi001, 2025-04-23:
> > I have made a fix to llvm's main branch, [llvm/llvm-project@b0524f3](https://github.com/llvm/llvm-project/commit/b0524f332958b6e593868533127fd0651bdcf553), could you please try if it works?
> > [@beakthoven](https://github.com/beakthoven)
> 
> I saw the PR, I'm currently outside travelling so I didn't respond. I'll test the patch as soon as I reach back (ETA 2 days).

I see. Sorry, I forgot that you are busy recently.