avrdudes/avr-libc #1023: man pages are not installed

matwey, 2025-02-21:
Hello, 

I am trying to compile avr-libc 2.2.1:

`./configure  --host=avr --build=`./config.guess` --enable-man-doc`

I see that the man pages are enabled:

```
checking for doxygen... yes
configure: Enabling PDF docs
configure: Enabling HTML docs
configure: Enabling man pages
```

When I do `make install` or `make install-data`, no man pages are installed.

sprintersb, 2025-02-21:
What happens with `make install-dox-man`?

matwey, 2025-02-21:
```
> make install-dox-man
make: *** No rule to make target 'install-dox-man'.  Stop.
```

sprintersb, 2025-02-21:
Running is from `$builddir/doc/api`? Like (cd $builddir/doc/api; make install-dox-man)`

matwey, 2025-02-21:
It is better now! However, it is still failed. I am not sure where `./dox_latex_header.tex` should come from:
```
> make install-dox-man
if test -n "/home/matwey/temp/avr-libc-2.2.1/avr/lib/avr4/exit.o /home/matwey/temp/avr-libc-2.2.1/avr/devices/atmega8/crtatmega8.o -lgcc" ; then \
	NODEFLIBS="-nostartfiles -nodefaultlibs" ; \
fi; \
cd ../../doc/examples/demo && \
	make DEFS="-I/home/matwey/temp/avr-libc-2.2.1/include"  \
		LIBS="$NODEFLIBS /home/matwey/temp/avr-libc-2.2.1/avr/lib/avr4/exit.o /home/matwey/temp/avr-libc-2.2.1/avr/devices/atmega8/crtatmega8.o -lgcc" all
make[1]: Entering directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/demo'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/demo'
for demo in demo largedemo stdiodemo; do \
  ( cd ../../doc/examples/$demo && \
  make dox ); \
done
make[1]: Entering directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/demo'
make[1]: Nothing to be done for 'dox'.
make[1]: Leaving directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/demo'
make[1]: Entering directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/largedemo'
make[1]: Nothing to be done for 'dox'.
make[1]: Leaving directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/largedemo'
make[1]: Entering directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/stdiodemo'
make[1]: Nothing to be done for 'dox'.
make[1]: Leaving directory '/home/matwey/temp/avr-libc-2.2.1/doc/examples/stdiodemo'
cat ./main_page.dox.in > main_page.dox
cat ../../LICENSE >> main_page.dox
echo '\endverbatim' >> main_page.dox
echo '*/' >> main_page.dox
doxygen doxygen.config
warning: Tag 'HTML_TIMESTAMP' at line 1194 of file 'doxygen.config' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
error: tag LATEX_HEADER: header file './dox_latex_header.tex' does not exist
Exiting...
make: *** [Makefile:1248: latex_src/refman.tex] Error 1
```

sprintersb, 2025-02-21:
That file was added in cdde8f3c2ed77ac85111bcf30c40e78
Though I am not at all familiar with all this, maybe Jörg has a clue...

matwey, 2025-02-21:
This file not in the release tarball. I guess it is a bug in `Makefile.am`.

matwey, 2025-02-22:
I have just checked that `make dist` produces the tarball capable for building `install-dox-man` correctly. Are you going to redo distribution tarballs or make new release?

sprintersb, 2025-02-22:
Currently there are no plans for a new release.

Though it would make sense to roll a new release around the time when GCC v15 is released (spring), as there are [new features](https://gcc.gnu.org/gcc-15/changes.html#avr) like `-mcvt` that require libc support.

sprintersb, 2025-07-25:
`make distckeck` should work again.

closed 2025-07-25 by sprintersb