avrdudes/avr-libc #1030: Potential bug in attiny814 libc (snprintf)?

isloven, 2025-04-08:
I'm testing avr-gcc 14.2.0 and avr-libc 2.2.0 on an attiny814 micro and I'm seeing unexpected behavior with snprintf (and other printf) functions.

Toggling IO pins and other operations on the micro work so it seems that the compiler is working correctly. 

Tests are compiled with -mmcu=attiny814:
avr-gcc -g -mmcu=attiny814 -o sprintf_test main.c

I am using a simple "uart_send_byte" function that has been tested and it works correctly. Example:

```
char buffer[16] = {'1', '2', '3', '4'};
int len = 4;

uart_send_byte(len & 0xFF);
uart_send_byte((len >> 8) & 0xFF);
uart_send_byte((len >> 16) & 0xFF);
uart_send_byte((len >> 24) & 0xFF);

for(int i=0; i<len; i++)
    uart_send_byte(buffer[i]);

```
Output on the UART terminal console (in hex) which is correct:
04 00 00 00 31 32 33 34

Here is a very simple example of the problem with snprintf:

```
char buffer[16] = {0};
long len = snprintf(buffer, sizeof(buffer), "%d", 1234);

uart_send_byte(len & 0xFF);
uart_send_byte((len >> 8) & 0xFF);
uart_send_byte((len >> 16) & 0xFF);
uart_send_byte((len >> 24) & 0xFF);

for(int i=0; i<len; i++)
    uart_send_byte(buffer[i]);
```

This produces the following on the UART (in hex):
04 00 00 00 F5 07 98 F4

So the length returned (4) seems correct, however the buffer contains garbage.

If I replace the snprintf line above with this:

`long len = snprintf(buffer, sizeof(buffer), "AB\n");`

The output looks like this (correct):
03 00 00 00 41 42 0A

However, if I add just one more character, it breaks:
`long len = snprintf(buffer, sizeof(buffer), "ABC\n");`

Output: 
04 00 00 00 8F 89 F4 01

I've also attempted to test this with the other vfprintf implementations by adding this to the compiler/linker and they also seem to be broken:
```
-Wl,-u,vfprintf -lprintf_min
-Wl,-u,vfprintf -lprintf_flt -lm
```

Any help would be greatly appreciated! Thanks!

sprintersb, 2025-04-09:
From my experience, the most common issue with this kind of devices is that someone isn't uploading `.rodata` to the device.

Anyway, since you didn't provide a complete test case, and I don't have the respectibe hardware, I gave AVRtest a try in order to narrow down the issue. I used the following test case:
```c
#include <stdint.h>
#include <stdio.h>

#include "avrtest.h"

__attribute__((noipa))
void uart_send_byte(uint8_t x)
{
    LOG_FMT_X8 (" %02x", x);
}

__attribute__((noipa))
void send (void)
{
    char buffer[16] = {0};
    long len = snprintf(buffer, sizeof(buffer), "%d", 1234);

    uart_send_byte(len & 0xFF);
    uart_send_byte((len >> 8) & 0xFF);
    uart_send_byte((len >> 16) & 0xFF);
    uart_send_byte((len >> 24) & 0xFF);

    for(int i=0; i<len; i++)
        uart_send_byte(buffer[i]);
}

__attribute__((noipa))
void send_ABC (void)
{
    char buffer[16] = {0};
    long len = snprintf(buffer, sizeof(buffer), "ABC\n");

    uart_send_byte(len & 0xFF);
    uart_send_byte((len >> 8) & 0xFF);
    uart_send_byte((len >> 16) & 0xFF);
    uart_send_byte((len >> 24) & 0xFF);

    for(int i=0; i<len; i++)
        uart_send_byte(buffer[i]);
}

int main (void)
{
    send ();
    LOG_STR ("\n");
    send_ABC ();
    LOG_STR ("\n");
    return 0;
}
```
The output is as expected:
```
$ avrtest-xmega -mmcu=avrxmega3 snp.elf
 04 00 00 00 31 32 33 34
 04 00 00 00 41 42 43 0a

 exit status: EXIT
      reason: exit 0 function called
     program: snp.elf
exit address: 00092a
total cycles: 3866
total instr.: 2697
```
To run the test case, follow these steps:
* Download AVRtest from https://github.com/sprintersb/atest to some folder `someplace`. Dependencies to build AVRtest are `make`, `avr-gcc` and a host `GCC` or `Clang`.
* Go to `someplace` and run `make`, then `make exit-attiny814.o`.
* Build the program from above with `avr-gcc -mmcu=attiny814 snp.c -o snp.elf -O0 -I someplace someplace/exit-attiny814.o`
* Simulate the program with `avrtest-xmega -mmcu=avrxmega3 snp.elf``.

isloven, 2025-04-11:
Your comment regarding not copying the .rodata section was very helpful! I copied the avr-objcopy command line from Atmel's Start project without paying attention to the flags and that was the issue.

So this is definitely not a bug and instead is a typical dummy user issue. 

Thanks you kindly for your help.

sprintersb, 2025-04-11:
So the question is why you are (still) using `objcopy` (and hence Intex Hex obviously) to begin with. Modern programmers like `avrdude` understand ELF and know what to upload for "flash".  One source of bugs less...

sprintersb, 2025-04-11:
Not a libc issue (user did't upload .rodata).

closed 2025-04-11 by sprintersb