avrdudes/avr-libc #1046: Packaging: FTBFS in Debian Testing

melizasw, 2025-09-06:
I've got another failure to build avr-libc on Debian related to generating the PDF reference manual. They just changed the version of texlive-base from `2024.20250309` to `2025.20250727` which is no doubt the cause of the problem.

This time I made sure it wasn't due to Debian specific packaging settings by running

```
./bootstrap
./configure --build=`./config.guess` --host=avr --prefix=/usr/lib --enable-device-lib  --disable-versioned-doc --enable-doc --datadir=/usr/share --bindir=/usr/bin --mandir=/usr/share/man
make
```

After a few minutes the build hangs with `pdflatex` running. Here is the output:

```
cd latex_src && make refman.pdf
Fixing transparency for avr-libc-logo.png
make[5]: Entering directory '/home/swm/workspace/debian/build/avr-libc/avr-libc_2.2.1/doc/api/latex_src'
rm -f *.ps *.dvi *.aux *.toc *.idx *.ind *.ilg *.log *.out *.brf *.blg *.bbl refman.pdf
pdflatex refman
Fixing transparency for bc_sd.png
Fixing transparency for bc_s.png
Fixing transparency for closed.png
Fixing transparency for demo.png
This is pdfTeX, Version 3.141592653-2.6-1.40.28 (TeX Live 2025/Debian) (preloaded format=pdflatex)
 restricted \write18 enabled.
Fixing transparency for malloc-std.png
Fixing transparency for malloc-x1.png
Fixing transparency for malloc-x2.png
entering extended mode
(./refman.tex
LaTeX2e <2025-06-01> patch level 1
L3 programming layer <2025-07-19>

Fixing transparency for nav_fd.png
Fixing transparency for nav_f.png
Fixing transparency for nav_g.png
Fixing transparency for nav_hd.png
Fixing transparency for nav_h.png
Fixing transparency for open.png
Fixing transparency for splitbard.png
Fixing transparency for splitbar.png
Fixing transparency for sync_off.png
Fixing transparency for sync_on.png
Fixing transparency for tab_ad.png
Fixing transparency for tab_a.png
Fixing transparency for tab_bd.png
Fixing transparency for tab_b.png
Fixing transparency for tab_hd.png
Fixing transparency for tab_h.png
Fixing transparency for tab_sd.png
Fixing transparency for tab_s.png
echo timestamp > fix_png
```

If I run `pdflatex refman.tex` then `refmen.log` ends with
```
! Missing # inserted in alignment preamble.
<to be read again> 
                   &
l.290 \end{longtabu}
                    
There should be exactly one # between &'s, when an
\halign or \valign is being set up. In this case you had
none, so I've put one in; maybe that will work.

! Missing # inserted in alignment preamble.
<to be read again> 
                   \cr 
l.290 \end{longtabu}
                    
There should be exactly one # between &'s, when an
\halign or \valign is being set up. In this case you had
none, so I've put one in; maybe that will work.

! LaTeX Error: Not allowed in LR mode.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.290 \end{longtabu}
                    
You've lost some text.  Try typing  <return>  to proceed.
If that doesn't work, type  X <return>  to quit.

! LaTeX Error: Not allowed in LR mode.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.290 \end{longtabu}
                    
You've lost some text.  Try typing  <return>  to proceed.
If that doesn't work, type  X <return>  to quit.

! Missing \endgroup inserted.
<inserted t
```

I'm considering dropping generation of the reference manual, but am open to ideas to resolve this.

melizasw, 2025-09-18:
The Debian bug has been reassigned to the `doxygen-latex` package as there are apparently multiple packages having issues, not just avr-libc.

sprintersb, 2025-09-19:
If it helps you, I can buld the PDF for you.